flashg1/TeslaSolarCharger #2: Possibility to combine with other solar production driven automations

kimme1024, 2023-09-12:
Hi!

First of all thanks for this! I'm getting delivery of my Model S today so everything is ready to charge the car efficiently now!

However I'm having a question related to this.

Currently I'm running a similar (but way more simple) automation that turns on/off my pool's heat pump based on excess solar energy.

However, as these 2 automations will be running at the same time from may to september they will interfere with each-other.

At this point the automation does the following:
When I'm having an excess of 1kW for 30s, the heatpump turns on.
When I'm having a shortage of 200W for 1m, the heatpump turns off.

However, when the Tesla will be charging dynamically, there won't be any extra energy for the heatpump.

Any ideas?

Thanks in advance!

flashg1, 2023-09-12:
Hi @kimme1024,

Good question.  Had been thinking about that too.  How do you manage all the scripts to work as one to share the available solar electricity.  You need an energy management system!  Not sure if we have one for Home Assistant.

kimme1024, 2023-09-12:
> Hi @kimme1024,
> 
> Good question. Had been thinking about that too. How do you manage all the scripts to work as one to share the available solar electricity. You need an energy management system! Not sure if we have one for Home Assistant.

Hi @flashg1,

At this point I only use the heat pump this way.
Before I installed my split unit HVAC I used to heat/cool some rooms with excess solar together with the heat pump.
To achieve this I have put some timers in to prioritize some devices...

Example:

Check for overproduction:
When overproduction is more than 1kW for 30sec --> activate heating room 1
When overproduction is more than 1kW for 1m --> check room 1 is heating / warm enough / window open --> activate heating room 2
When overproduction is more than 1kW for 1m30s --> check room 1 is heating / warm enough / window open, check room 2 is heating / warm enough / window open --> activate heat pump pool

It's a bit clumsy but it works as it should. However nothing above works dynamically as your automation as they all constantly draw about 1kW a piece.

I think a possible solution is to add a feature to your automation that leaves room to prioritize another device/switch for a certain amount of energy. So if the Tesla draws 10A and I select to give priority to my heatpump, it lowers the current the Tesla draws needed for the other device and activates the switch for the heatpump. Then the Tesla modulates again with the excess energy as before.

It should only make a difference when it becomes cloudy and sunny again so it'll have to restart everything.

I hope this makes sense...

broodro0ster, 2023-09-14:
Same issue here when I'm cooling our summer with excess solar via our heatpump.
I haven't tried this but you could say this: if the car is charging with more than 3kW and there is less than 200W grid usage, start the heatpump for the pool. The heatpump will start and the tesla will start charging a little slower after 1 minute. 
I would only increase the stop script for the heatpump to 3 minutes before turning it off. Because the Tesla script needs some time to adapt.

kimme1024, 2023-09-14:
> Same issue here when I'm cooling our summer with excess solar via our heatpump.
> I haven't tried this but you could say this: if the car is charging with more than 3kW and there is less than 200W [...]

nice workaround! i'll look into it! but first we'll have to solve issue #1 :)

kimme1024, 2023-09-14:
Been tinkering a bit with it and I think following should work:

```
alias: Warmtepomp Zwembad Aan (icm Tesla)
description: ""
trigger:
  - platform: numeric_state
    entity_id: sensor.net_usage
    above: 1000
    for:
      hours: 0
      minutes: 0
      seconds: 30
  - platform: state
    entity_id:
      - device_tracker.wattson_location_tracker
    from: home
    to: not_home
  - platform: numeric_state
    entity_id: number.wattson_charging_amps
    above: 5
condition:
  - condition: or
    conditions:
      - condition: numeric_state
        entity_id: sensor.net_usage
        above: 1000
      - condition: numeric_state
        entity_id: number.wattson_charging_amps
        above: 5
      - condition: state
        entity_id: device_tracker.wattson_location_tracker
        state: not_home
action:
  - service: switch.turn_on
    data: {}
    target:
      entity_id: switch.zwembad
mode: single

```

The off-automation has a delay of 1min as the tesla automation refreshes every 30sec.

Works for now but I'll have to monitor it while it's running.

Thanks for the tip @broodro0ster

flashg1, 2024-11-17:
Closing case unless this is still an issue.

closed 2024-11-17 by flashg1