flashg1/TeslaSolarCharger #6: Energy usage between end of charging via solar and end of automation due to sunset

kimme1024, 2023-09-15:
Hi,

Today was a flawless day! So thanks again for this script!
However yesterday and today I noticed that while I wasn't charging (0 amps) there was still about 400w of unexplainable usage that I normally never have.

So before I start measuring that specific plug I'd like to ask that there's some kind of idle usage between the end of the solar charge and the end of the automation 30min before sunset?

Screenshots below:

Here you can see that I'm consuming 843w. At this point the tesla was charging at 0 amps, so not charging.
[image]

This was after the script ended. I went back to my normal idle use of around 400-450w (you can also see it clearly in the graph between midnight and 6 in the morning.
[image]

It's obviously not that much of energy that's been wasted but if it can be avoided it's always better :)

flashg1, 2023-09-16:
Nice charts you have there!

Not sure about the charger consumption when it is supplying 0 amps, but I doubt it is using up 400W.  I suspect you might have other appliances on at the time that you might not be aware of?

When charging from solar panels, please note there is a hard coded skew setting as commented in the script.  This is to allow slight fluctuation around 0 net grid power.

          # all_power_net < -69W, ie. need to increase charger current.
          # all_power_net >= 161W, ie. need to decrease charger current.

kimme1024, 2023-09-16:
> Nice charts you have there!
>
> Not sure about the charger consumption when it is supplying 0 amps, but I doubt it is using up 400W. I suspect you might have other appliances on at the time that you [...]

Hi, I'll try to test it again in the future, however there isn't that much nice weather coming up to test it with...
Anyway I don't have any clear idea of which device might cause this usage. It's also very clear (2 days in a row exactly the same) that the usage is between the amps get to 0 due to not enough solar and the script ending 30min before sunset. When the conditions are ideal I'll try to test and prove this using a smart socket with power metering so I can see the exact energy usage on that moment.

To be continued :)

kimme1024, 2023-09-17:
Did some testing as the same happens in the morning before there’s solar production.

When charger is on and charging at 0 amps there’s still a power consumption of around 400w. When I disable charging it stops immediately.

Could this be fixed by stopping the charge when there isn’t enough solar production? During a sunny day it isn’t an issue, however in rainy days it will consume a lot of unnecessary energy without charging the car.

[image]
Above you can see my normal draw first. Then a little hiccup with another automation, however when I fixed that the power consumption remained higher. When I stopped the charging completely you can see that my power consumption went back to normal.

flashg1, 2023-09-26:
I'm against turning off the charger for the following reasons,

1)  I against turning on and off the switch frequently to minimise electrical wear and tear.
2)  It complicates matter because the Tesla car can also turn off the charger on reaching charge limit or just below charge limit, in which case I am using that as a signal to terminate the script.

kimme1024, 2023-09-26:
> I'm against turning off the charger for the following reasons,
> 
> 1. I am against turning on and off the switch frequently to minimise electrical wear and tear.
> 2. It complicates matter because the Tesla car can also turn off the charger on reaching charge limit or just below charge limit, in which case I am using the charger off signal to terminate the script in addition to the charge limit trigger.

Ok. Would there be a way to set the minimum amps at 1 or 2 then?
When it’s on 0 I’m consuming about 400w anyway without charging. So the it’ll be better to put those 400w into the battery instead of wasting it to wherever it goes…

flashg1, 2023-09-28:
Added option to set the minimum charge current.  The default is 0A, but you can set it to a different number when setting up the Blueprint.  Please give it a try.

kimme1024, 2023-09-28:
> Added option to set the minimum charge current. The default is 0A, but you can set it to a different number when setting up the Blueprint. Please give it a try.

Thanks man! Installed it and it works for now. Will have to wait for tonight to see if it stays at 1A. :)

kimme1024, 2023-10-09:
Very weird thing today:

When I was making dinner this evening, the automation went from 0A to 5A when I was already pulling about 2kW from the grid. Therefore I had a 15min peak of 1150w higher than it should have been (which is "fined" here in Belgium since the beginning of this year).

```
tesla_solar_charger_script_scene  gewijzigd in 9 oktober 2023 om 18:02 getriggerd door automatisering Tesla Solar Charger  getriggerd door status van Wattson Charger
18:02:40 - 2 uur geleden

Wattson Charging amps  gewijzigd in 5 getriggerd door automatisering Tesla Solar Charger  getriggerd door status van Wattson Charger
18:02:40 - 2 uur geleden

Wattson Force data update  gewijzigd in 9 oktober 2023 om 18:02 getriggerd door automatisering Tesla Solar Charger  getriggerd door status van Wattson Charger
18:02:45 - 2 uur geleden
```

It happened exactly at the time that the charging switch should have been turned off by the script due to the sunset trigger.
Normally it turns off the switch and sets the charging amps to 5. This time it did the same, without turning off the switch.

flashg1, 2024-11-17:
Closing case unless this is still an issue.

closed 2024-11-17 by flashg1