flashg1/TeslaSolarCharger #7: Script working when plugging in after sunset

kimme1024, 2023-09-16:
Hi,

Another day, another question ;)

Today I've been charging using the script which worked flawlessly.
However this afternoon I left and I didn't come home before sunset.
When I plugged in the car, it started charging as expected, however the script didn't stop the charging.
I had to manually restart the automation for it to start working again.

Maybe there's a possibility to add an extra trigger on plugging in the charger that will check the sun's status so it'll be able to work as intended?

kimme1024, 2023-09-25:
Today I think I had an issue related to this.
I'm currently fixing this issue (https://github.com/flashg1/TeslaSolarCharger/issues/6) by switching on `stop charging` for the time between low solar production and sunset.
After sunset I'm turning off the switch so the script can work the next morning.
However yesterday I left and came back after sunset so for that I left on the `stop charging` switch.

After I came home I didn't restart the script, but I just disabled `stop charging`.

This morning, the script started, set the amps on 12 (while not having excess solar) and stopped the script after 125sec. As I only spotted this issue at 10 I've been pulling about 4kWh in peak tariff. Not that big of an issue if it happens once but as we also get "fined" for high peak usage on a 15min basis this can also result in bigger costs over time.

I don't know what caused the script to stop but i think it was this:
```
Exiting script due to charger turned off manually or automatically by car when charge limit reached.
```
However, none of the above was true.

Extra info:

```
Triggered by the sunrise with offset at 25 september 2023 om 08:02:11

Confirm Wattson Location tracker is Thuis testen

Confirm Wattson Charger is Ingeplugd testen

Service "Knop: Druk" aanroepen op "Wattson Wake up"

(button.wattson_wake_up) turned 2023-09-25T06:02:11.757697+00:00

Vertraging gebaseerd op een sjabloon
40 seconden later

Service "Knop: Druk" aanroepen op "Wattson Force data update"

(button.wattson_force_data_update) turned 2023-09-25T06:02:51.965709+00:00
11 seconden later

Vertraging gebaseerd op een sjabloon
5 seconden later

Service "Scène: Aanmaken" aanroepen op ""

Vertraging gebaseerd op een sjabloon
5 seconden later

Actie terwijl 1 voorwaarde overeenkomt herhalen

unknown

(number.wattson_charging_amps) turned 6
6 seconden later

(switch.wattson_charger) turned on
11 seconden later

(button.wattson_force_data_update) turned 2023-09-25T06:03:30.319674+00:00
35 seconden later

unknown

(scene.tesla_solar_charger_script_scene) turned 2023-09-25T06:04:05.763918+00:00
(number.wattson_charging_amps) turned 12
5 seconden later

(button.wattson_force_data_update) turned 2023-09-25T06:04:11.540289+00:00

Finished at 25 september 2023 om 08:04:17 (runtime: 125.58 seconds)
```
[image]

[image]

flashg1, 2023-09-26:
> When I plugged in the car, it started charging as expected, however the script didn't stop the charging.
> I had to manually restart the automation for it to start working again.

This is due Tesla polling issue, ie. HA is not in sync with Tesla car status.  I mentioned about 2 possible workarounds on github.  The easiest and sure way is to just manually run the script which you have done.

kimme1024, 2023-09-26:
Ok I’ll keep that in mind thanks!

closed 2023-09-26 by kimme1024