flashg1/TeslaSolarCharger #8: Automation stopped without clear reason

kimme1024, 2023-09-26:
This morning I noticed that the script didn't change the amps anymore. I checked and it did stop working indeed. However I've been looking around and it had no clear reason why it did.
Maybe it's a good idea to send a notification to the user's phone when the car stops charging unexpectedly?

Trace Timeline:

```
Triggered by the sunrise with offset at 26 september 2023 om 08:03:46

Confirm Wattson Location tracker is Thuis testen

Confirm Wattson Charger is Ingeplugd testen

Service "Knop: Druk" aanroepen op "Wattson Wake up"

(button.wattson_wake_up) turned 2023-09-26T06:03:46.276675+00:00

Vertraging gebaseerd op een sjabloon
40 seconden later

Service "Knop: Druk" aanroepen op "Wattson Force data update"

(button.wattson_force_data_update) turned 2023-09-26T06:04:26.443529+00:00

Vertraging gebaseerd op een sjabloon
5 seconden later

Service "Scène: Aanmaken" aanroepen op ""

Vertraging gebaseerd op een sjabloon
5 seconden later

Actie terwijl 1 voorwaarde overeenkomt herhalen

unknown

(number.wattson_charging_amps) turned 6
5 seconden later

(switch.wattson_charger) turned on
11 seconden later

(button.wattson_force_data_update) turned 2023-09-26T06:04:53.602238+00:00
37 seconden later

unknown

(number.wattson_charging_amps) turned 5
1 seconde later

(number.wattson_charging_amps) turned 4
1 seconde later

(number.wattson_charging_amps) turned 3
(number.wattson_charging_amps) turned 2
(number.wattson_charging_amps) turned 1
1 seconde later

(number.wattson_charging_amps) turned 0
(button.wattson_force_data_update) turned 2023-09-26T06:05:36.503737+00:00
36 seconden later

unknown

(button.wattson_force_data_update) turned 2023-09-26T06:06:12.295027+00:00
37 seconden later

unknown

(button.wattson_force_data_update) turned 2023-09-26T06:06:49.125367+00:00
36 seconden later
[... 968 lines cut ...]

unknown

(button.wattson_force_data_update) turned 2023-09-26T07:56:30.101616+00:00
35 seconden later

unknown

(button.wattson_force_data_update) turned 2023-09-26T07:57:05.569347+00:00
36 seconden later

unknown

(button.wattson_force_data_update) turned 2023-09-26T07:57:42.056310+00:00
35 seconden later

unknown

(number.wattson_charging_amps) turned 9
(button.wattson_force_data_update) turned 2023-09-26T07:58:18.076839+00:00

Stopped because an error was encountered at 26 september 2023 om 09:58:18 (runtime: 6871.90 seconds)

```
[image]

flashg1, 2023-09-26:
Please check the HA system log around that time.  There should be more clues there as to the cause, ie.

Settings > System > Logs

Suspect the Tesla API call failed or timed out.  If that is the case, not sure if anything can be done here unless there is a way to capture that exception.

kimme1024, 2023-09-26:
> Please check the HA system log around that time. There should be more clues there as to the cause, ie.
> 
> Settings > System > Logs
> 
> Suspect the Tesla API call failed or timed out. If that is the case, not sure if anything can be done here unless there is a way to capture that exception.

I’ll check this when I’m back at home.

kimme1024, 2023-09-26:
This came up in the logs. Any useful info in here?

```
Logger: homeassistant.components.automation.tesla_solar_charger
Source: custom_components/tesla_custom/base.py:82 
Integration: Automatisering ([documentation](https://www.home-assistant.io/integrations/automation), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+automation%22)) 
First occurred: 09:58:18 (3 occurrences) 
Last logged: 09:58:18

Tesla Solar Charger: Repeat at step 7: Error executing script. Unexpected error for call_service at pos 4:
Tesla Solar Charger: Error executing script. Unexpected error for repeat at pos 7:
While executing automation automation.tesla_solar_charger
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 704, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 666, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1969, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2006, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 124, in _async_press_action
    await self.async_press()
  File "/config/custom_components/tesla_custom/button.py", line 84, in async_press
    await self.update_controller(wake_if_asleep=True, force=True)
  File "/config/custom_components/tesla_custom/base.py", line 82, in update_controller
    await self.coordinator.controller.update(
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 757, in update
    cars = await self.get_vehicles()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 301, in get_vehicles
    return (await self.api("VEHICLE_LIST"))["response"]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 1348, in api
    return await self.__post_with_retries_except_unavailable(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 88, in async_wrapped
    return await fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 47, in __call__
    do = self.iter(retry_state=retry_state)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/__init__.py", line 314, in iter
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/_base.py", line 449, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 50, in __call__
    result = await fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 1380, in __post_with_retries_except_unavailable
    return await self.__connection.post(command, method=method, data=data, url=url)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/connection.py", line 162, in post
    return await self.__open(url, method=method, headers=self.head, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/connection.py", line 213, in __open
    raise TeslaException(resp.status_code)
teslajsonpy.exceptions.TeslaException
```

kimme1024, 2023-09-26:
More logs:

```
2023-09-26 09:58:18.114 ERROR (MainThread) [homeassistant.components.automation.tesla_solar_charger] Tesla Solar Charger: Repeat at step 7: Error executing script. Unexpected error for call_service at pos 4: 
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 704, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 666, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1969, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2006, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 124, in _async_press_action
    await self.async_press()
  File "/config/custom_components/tesla_custom/button.py", line 84, in async_press
    await self.update_controller(wake_if_asleep=True, force=True)
  File "/config/custom_components/tesla_custom/base.py", line 82, in update_controller
    await self.coordinator.controller.update(
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 757, in update
    cars = await self.get_vehicles()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 301, in get_vehicles
    return (await self.api("VEHICLE_LIST"))["response"]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 1348, in api
    return await self.__post_with_retries_except_unavailable(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 88, in async_wrapped
    return await fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 47, in __call__
    do = self.iter(retry_state=retry_state)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/__init__.py", line 314, in iter
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/_base.py", line 449, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 50, in __call__
    result = await fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 1380, in __post_with_retries_except_unavailable
    return await self.__connection.post(command, method=method, data=data, url=url)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/connection.py", line 162, in post
    return await self.__open(url, method=method, headers=self.head, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/connection.py", line 213, in __open
    raise TeslaException(resp.status_code)
teslajsonpy.exceptions.TeslaException
2023-09-26 09:58:18.140 ERROR (MainThread) [homeassistant.components.automation.tesla_solar_charger] Tesla Solar Charger: Error executing script. Unexpected error for repeat at pos 7: 
[... 162 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 47, in __call__
    do = self.iter(retry_state=retry_state)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/tenacity/__init__.py", line 314, in iter
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/_base.py", line 449, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.11/site-packages/tenacity/_asyncio.py", line 50, in __call__
    result = await fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/controller.py", line 1380, in __post_with_retries_except_unavailable
    return await self.__connection.post(command, method=method, data=data, url=url)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/connection.py", line 162, in post
    return await self.__open(url, method=method, headers=self.head, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/teslajsonpy/connection.py", line 213, in __open
    raise TeslaException(resp.status_code)
teslajsonpy.exceptions.TeslaException
```

kimme1024, 2023-09-28:
Happened again today... Could there be a possibility to add a failsafe in the script for when this happens?
At this point I can't seem to trust the script. It worked flawlessly yesterday but failed again today...

```
stop: >-
  Exiting script due to charger turned off manually or automatically by car when
  charge limit reached.
```

```
Triggered by the sunrise with offset at 28 september 2023 om 08:06:55

Confirm Wattson Location tracker is Thuis testen

Confirm Wattson Charger is Ingeplugd testen

Service "Knop: Druk" aanroepen op "Wattson Wake up"

(button.wattson_wake_up) turned 2023-09-28T06:06:55.863864+00:00

Vertraging gebaseerd op een sjabloon
40 seconden later

Service "Knop: Druk" aanroepen op "Wattson Force data update"

(button.wattson_force_data_update) turned 2023-09-28T06:07:36.057267+00:00

Vertraging gebaseerd op een sjabloon
5 seconden later

Service "Scène: Aanmaken" aanroepen op ""

Vertraging gebaseerd op een sjabloon
5 seconden later

Actie terwijl 1 voorwaarde overeenkomt herhalen

unknown

(number.wattson_charging_amps) turned 6
5 seconden later

(switch.wattson_charger) turned on
11 seconden later

(button.wattson_force_data_update) turned 2023-09-28T06:08:03.298540+00:00
36 seconden later

unknown

(scene.tesla_solar_charger_script_scene) turned 2023-09-28T06:08:39.327585+00:00
(number.wattson_charging_amps) turned 5
5 seconden later

(button.wattson_force_data_update) turned 2023-09-28T06:08:44.669361+00:00

Finished at 28 september 2023 om 08:08:50 (runtime: 114.23 seconds)
```

kimme1024, 2023-09-29:
Happened today as well…

flashg1, 2023-09-29:
Just wondering what could be turning off your charger?

What did you set your battery charge limit to?

At what battery percentage did it stop charging?

kimme1024, 2023-09-29:
> Just wondering what could be turning off your charger?
> 
> What did you set your battery charge limit to?
> 
> At what battery percentage did it stop charging?

Yesterday at 74 with limit of 80
Today at 78 with limit of 90

flashg1, 2023-09-29:
Do you have scheduled charging set for your Tesla?  Ie. Do you have charging start and stop time programmed into your Tesla?

flashg1, 2023-09-29:
Then I don't know.  The issue is most likely environmental. Something turned off your charger.  The only way I might be able to help you is if the problem is reproducible and I have the exact reproduction steps and settings.

kimme1024, 2023-09-29:
> Then I don't know. The issue is most likely environmental. Something turned off your charger. The only way I might be able to help you is if the problem is reproducible and I have the exact reproduction steps and settings.

Anything you need from me to figure this out?

flashg1, 2023-09-30:
There is only one place in the script where the charger is turned off, and that is at end of script just before exit.

The message showed the end of script was never reached and that the charger was turned off external to the automation.

I suggest you check if you have any other Tesla automations running that can turn off the charger.

You will get the same message is you click "Stop charging" in the Tesla app.

kimme1024, 2023-09-30:
It happened again today.
What’s weird is that it now happens very quickly after the automation starts. This is new since the updated version with the minimum charging amps. Before it took longer to crash and it didn’t crash everyday at the same moment.

kimme1024, 2023-09-30:
It only does one cycle. Then it stops.
```

Triggered by the sunrise with offset at 30 september 2023 om 08:40:06

Confirm Wattson Location tracker is Thuis testen

Confirm Wattson Charger is Ingeplugd testen

Service "Knop: Druk" aanroepen op "Wattson Wake up"

(button.wattson_wake_up) turned 2023-09-30T06:40:06.208828+00:00

Vertraging gebaseerd op een sjabloon
40 seconden later

Service "Knop: Druk" aanroepen op "Wattson Force data update"

(button.wattson_force_data_update) turned 2023-09-30T06:40:46.413710+00:00

Vertraging gebaseerd op een sjabloon
5 seconden later

Service "Scène: Aanmaken" aanroepen op ""

Vertraging gebaseerd op een sjabloon
5 seconden later

Actie terwijl 1 voorwaarde overeenkomt herhalen

unknown

(number.wattson_charging_amps) turned 6
5 seconden later

(switch.wattson_charger) turned on
11 seconden later

(button.wattson_force_data_update) turned 2023-09-30T06:41:13.548043+00:00
37 seconden later

unknown

(scene.tesla_solar_charger_script_scene) turned 2023-09-30T06:41:50.195452+00:00
(number.wattson_charging_amps) turned 5
5 seconden later

(button.wattson_force_data_update) turned 2023-09-30T06:41:55.556283+00:00

Finished at 30 september 2023 om 08:42:01 (runtime: 115.03 seconds)

```

[image]

kimme1024, 2023-10-01:
Exactly the same today. Everytime after 114/115 seconds after running the script with sunrise as trigger. When I manually start the script afterwards, there’s no problem.
As the issue is very “reliable” I really think it’s caused by the script. The reliability was introduced since the last update (minimal charging amps feature).

flashg1, 2024-11-17:
Closing case unless this is still an issue.

closed 2024-11-17 by flashg1