flashg1/TeslaSolarCharger #9: Script doesn't stick to the amp limit

kimme1024, 2023-10-10:
This morning somehow the script stopped adapting the amps.
In the log below you can see that the script changed the amps at 10:15, did a couple of refreshes afterwards but nothing past that until I restarted the automation at 11:46.

```
Wattson Force data update  gewijzigd in 10 oktober 2023 om 10:14 getriggerd door automatisering Tesla Solar Charger  triggered
10:14:57 - 2 uur geleden

Wattson Charging amps  gewijzigd in 8 getriggerd door automatisering Tesla Solar Charger  triggered
10:15:34 - 2 uur geleden

Wattson Force data update  gewijzigd in 10 oktober 2023 om 10:15 getriggerd door automatisering Tesla Solar Charger  triggered
10:15:34 - 2 uur geleden

Wattson Force data update  gewijzigd in 10 oktober 2023 om 10:16 getriggerd door automatisering Tesla Solar Charger  triggered
10:16:10 - 2 uur geleden

Wattson Force data update  gewijzigd in 10 oktober 2023 om 10:16 getriggerd door automatisering Tesla Solar Charger  triggered
10:16:46 - 2 uur geleden

Wattson Force data update  gewijzigd in 10 oktober 2023 om 10:17 getriggerd door automatisering Tesla Solar Charger  triggered
10:17:21 - 2 uur geleden
```

[image]

After restarting the automation I now noticed that the car is charging at 13A while the limit in de script is set to 12A (due to the issue mentioned in https://github.com/flashg1/TeslaSolarCharger/issues/4.

[image]
[image]

In 2 weeks I'm getting my MCU1 replaced by the newer MCU2. After that we can rule out the fact that the issue lies there.

flashg1, 2023-10-10:
Please also consider including the script "Running instance count" in your control GUI to give you instant feedback that the script has been triggered.  1=script is running, 0=script not running.

https://github.com/flashg1/TeslaSolarCharger#dashboard-tesla-solar-charger-control

  - entity: automation.[YourTeslaName]_solar_charger_automation
  - type: attribute
    entity: automation.[YourTeslaName]_solar_charger_automation
    attribute: current
    name: Running instance count
  - type: attribute
    entity: automation.[YourTeslaName]_solar_charger_automation
    attribute: last_triggered
    name: Last triggered

flashg1, 2023-10-10:
This showed the charging current was not set on startup, ie. charger was already turned on and charging at whatever current before you started the script.  This is to allow the "Charge from grid" feature.  The script will only change the current when there is a change in the available solar electricity.

[image]

kimme1024, 2023-10-10:
> This showed the charging current was not set on startup, ie. charger was already turned on and charging at whatever current before you started the script. This is to allow the "Charge from grid" [...]

ah okay... I think I know what went wrong then...
To solve the issue where the script stopped as in https://github.com/flashg1/TeslaSolarCharger/issues/8, I made an automation that starts the script again 5min after the original script starts because when it failed, it always did after 114 seconds.
When the script didn't stop, my automation simply didn't do anything (I assumed) as it stopped with an error saying that the script was already running.
This might have caused the issue today, however it's weird that it didn't do the same everyday.

flashg1, 2024-11-17:
Closing case unless this is still an issue.

closed 2024-11-17 by flashg1