flashg1/TeslaSolarCharger #10: Home Battery - Function?

shagadeli, 2023-10-16:
I want to thank you for this blueprint, really appreciated.
This is not really an issue, just a question regarding the home battery. I looked at the yaml but was unsure of what use does the automation make of the home battery charge/discharge power and current SoC.

I have these values feeded into the automation, and would be great do make some use of it. For example, at late afternoon I would prefer that the solar surplus was feeded into the home battery instead of the car. The same would be true for early morning.

Is this the purpose of adding this data to the automation?
Thanks!

flashg1, 2023-10-26:
Sorry battery is not supported at the moment. I left it there as a place holder for future enhancement.

shagadeli, 2023-10-26:
Thanks for the clarification!

closed 2023-10-26 by shagadeli