flashg1/TeslaSolarCharger #12: Feature Request: Limit charging from grid based on realtime grid usage

kimme1024, 2023-11-18:
During the winter months it's sadly sometimes needed to charge your car from the grid.
As I'm having peak tariff I have to do this at night.
However, in Belgium, we're also fined on our peak quarterly usage. This morning, the car was charging, I was still asleep and my wife used the oven, therefore our peak usage for this month raised a lot. (The highest 15min of the month will count)

To avoid this it might be useful to implement the opposite of the script for the case when it's charging from the grid.

Example:

When charging from grid, you can set a grid limit (ex 4000W). So the car can charge at it's maximum current, but when another device starts using electricity which causes the grid usage to exceed the limit, it lowers the car's current for the time that this occurs.

I think this would be a very useful addition to this project!

ddeconin-gh, 2024-06-13:
+1 for this request :-)

kimme1024, 2024-10-12:
Yesterday I've had a genius friend come over and he quickly adapted the script to achieve this request.
It's done in a couple of hours so probably not the cleanest, nor most efficient but after testing it seemed to work just fine.

For those who want to use, further develop it, or implement it in TeslaSolarCharger, you can find it here: https://github.com/kimme1024/TeslaNightCharger
**Please note that it's VERY experimental and barely tested at this point.**

As it's currently a separate script you have to run a couple automations to switch between the scripts when using it. The following seems to work for now:

```
alias: Tesla Grid Charge AAN
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.tesla_model_s_charge_from_grid
    from: "off"
    to: "on"
condition: []
action:
  - action: automation.turn_off
    metadata: {}
    data:
      stop_actions: false
    target:
      entity_id: automation.tesla_solar_charger
  - action: automation.turn_off
    metadata: {}
    data:
      stop_actions: true
    target:
      entity_id: automation.turn_off_wattson_polling
  - delay:
      hours: 0
      minutes: 2
      seconds: 0
      milliseconds: 0
  - action: switch.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: switch.wattson_polling
  - action: automation.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: automation.tesla_night_charger_automation
  - action: automation.trigger
    metadata: {}
    data:
      skip_condition: true
    target:
      entity_id: automation.tesla_night_charger_automation
mode: single

```

```
alias: Tesla Grid Charge UIT
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.tesla_model_s_charge_from_grid
    from: "on"
    to: "off"
condition: []
action:
  - metadata: {}
    data: {}
    target:
      entity_id: input_boolean.tesla_model_s_charge_from_grid
    action: input_boolean.turn_off
  - action: automation.turn_off
    metadata: {}
    data:
      stop_actions: false
    target:
      entity_id: automation.tesla_night_charger_automation
  - action: automation.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: automation.turn_off_wattson_polling
  - delay:
      hours: 0
      minutes: 5
      seconds: 0
      milliseconds: 0
  - metadata: {}
    data: {}
    target:
      entity_id: switch.wattson_charger
    action: switch.turn_off
  - action: automation.turn_on
    metadata: {}
    data: {}
    target:
      entity_id: automation.tesla_solar_charger
  - action: automation.trigger
    metadata: {}
    data:
      skip_condition: true
    target:
      entity_id: automation.tesla_solar_charger
mode: single

```

**DISCLAIMER**
Use this at your own risk, also as I didn't make this myself I can't give any support on this. Look at it as a bump in the right direction for someone with the right knowledge to further develop it.

flashg1, 2024-11-14:
> During the winter months it's sadly sometimes needed to charge your car from the grid. As I'm having peak tariff I have to do this at night. However, in Belgium, we're also fined on our peak [...]

This might do what you want.  Please try v2.0 and set the daytime/nighttime power offsets to a negative value, eg. -10000 to draw a maximum of 10000 watts from grid.  Please don't forget to turn on charge from grid.

closed 2025-02-15 by flashg1