flashg1/TeslaSolarCharger #13: Starts charging in the morning and stops adapting the amps immediately

kimme1024, 2023-11-23:
Lately the script starts every morning as planned. Sets the charging amps at 5 but as it’s winter there’s no sun. The script doesn’t change the amps and quits, but leaves the car charging at 5A.
Is there a way to fix this?

```
Triggered by the sunrise with offset at 23 november 2023 om 08:40:23

Confirm Wattson Location tracker is Thuis testen

Confirm Wattson Charger is Ingeplugd testen

Service "Knop: Druk" aanroepen op "Wattson Wake up"

(button.wattson_wake_up) turned 2023-11-23T07:40:23.786591+00:00
11 seconden later

Vertraging gebaseerd op een sjabloon
40 seconden later

Service "Knop: Druk" aanroepen op "Wattson Force data update"

(button.wattson_force_data_update) turned 2023-11-23T07:41:14.351179+00:00

Vertraging gebaseerd op een sjabloon
5 seconden later

Service "Scène: Aanmaken" aanroepen op ""

Vertraging gebaseerd op een sjabloon
5 seconden later

Actie terwijl 1 voorwaarde overeenkomt herhalen

unknown

(number.wattson_charging_amps) turned 6
5 seconden later

(switch.wattson_charger) turned on
11 seconden later

(button.wattson_force_data_update) turned 2023-11-23T07:41:41.420881+00:00
36 seconden later

unknown

(scene.tesla_solar_charger_script_scene) turned 2023-11-23T07:42:17.335513+00:00
(number.wattson_charging_amps) turned 5
5 seconden later

(button.wattson_force_data_update) turned 2023-11-23T07:42:22.657680+00:00

Finished at 23 november 2023 om 08:42:28 (runtime: 124.36 seconds)
```

Kanishkaz, 2024-11-11:
I'm having a similar issue. The integration stops after starting the charge in the morning after sunrise.
@kimme1024 did you find a solution?

kimme1024, 2024-11-12:
> I'm having a similar issue. The integration stops after starting the charge in the morning after sunrise. @kimme1024 did you find a solution?

Hi,

I managed to "fix" it with a workaround with 2 automations that make sure the script is active in the morning and the evening to start and stop successfully. 

Morning (3min after script starts up):

```
alias: "Tesla ochtend "
description: ""
trigger:
  - platform: sun
    event: sunrise
    offset: 0
condition: []
action:
  - delay:
      hours: 0
      minutes: 58
      seconds: 0
      milliseconds: 0
  - target:
      entity_id: automation.tesla_solar_charger
    data:
      skip_condition: true
    action: automation.trigger
mode: single
```

Evening (3min before script ends):

```
alias: Tesla avond
description: ""
trigger:
  - platform: sun
    event: sunset
    offset: "-00:57:00"
condition: []
action:
  - target:
      entity_id: automation.tesla_solar_charger
    data:
      skip_condition: true
    action: automation.trigger
mode: single
```

closed 2025-02-15 by flashg1