flashg1/TeslaSolarCharger #14: Dynamic pricing charging

patatman, 2023-12-11:
Hi there, and thanks for creating this awesome blueprint.

Lately I've switched to a dynamic contract, and would love to add the possibility to charge the car when the price is at a certain point. 
I'm thinking about adding it to your blueprint, and setting a PR. But I'm not sure if this messes to much with your original intent of this blueprint.

Let me know if your open to this change, or if I can better fork this into my own blueprint.

patatman, 2023-12-11:
Thinking about this a bit longer, and it might be easier to create a new automation that:
Trigger: When hourly price changes

**Choose 1:**
_Condition:_ When price is below or equal to set maximum
_Action:_ Set charge amps to max
_Action:_ toggles the "charge from grid" switch on.
_Action:_ Trigger your script. 

**Choose 2:**
_Condition:_ When price is above to set maximum
_Action:_ toggles the "charge from grid" switch off.

flashg1, 2024-11-17:
You need price triggers.  Better to be a separate script.  Closing case.

closed 2024-11-17 by flashg1