flashg1/TeslaSolarCharger #15: 0A equal about 690W usage Tesla Gen 3 Wall charger

LasseBang, 2024-01-08:
Hi,

Love this code/project.

But one issue I face is when not enough excess solar power the car use about 690-720W at 0A, guess it how much it is to keep car alive :-)

Could it be possible when it hits 0A to turn of charger switch?

kimme1024, 2024-02-07:
I already mentioned this before.
Turning off the switch would wear out the relay too fast, instead you can do the following:

- set the minimum charging amps in the automation to 1. Then it uses only 230w.
- When you know there will be no solar energy for a longer period, just turn on the switch “stop charging” and re-enable it when there’s more sun.

LasseBang, 2024-02-07:
> I already mentioned this before. Turning off the switch would wear out the relay too fast, instead you can do the following:
>
> * set the minimum charging amps in the automation to 1. Then it uses [...]

That make sense :-) yea but I use minimum of 690W at 1A because it 3 phase

closed 2024-03-25 by LasseBang