flashg1/TeslaSolarCharger #18: Charger Voltage variable and its description is not the correct for its purpose

PVi1, 2024-05-26:
Hi @flashg1,

thank you for this blueprint, it had saved me a lot of time and motivated me to better understand HA automation!

In my case I am using it with Tesla Model S and Tesla Wall Connector gen 3 in the EU.

Issue:
Charger Voltage variable and its description is misleading. 
Initially I have set it to 400V and found out that automation always overshoot and started to oscillate.
[image]

In the same time, this was the FVE output:
[image]

Based on the [Tesla](https://www.tesla.com/en_EU/support/charging/onboard-charger) in the EU we have 400V Wye grid.

Based on the [this post](https://teslamotorsclub.com/tmc/threads/three-phase-charging-question.262716/#post-6581602)  Tesla in the EU uses 230V built in/onboard car charger and even in case when Tesla Wall Connector is used, 230V is used to charge the car. On the other side the car is able to stack/combine the energy from phase1, phase2 and phase3 aka (L1, L2, L3). Moreover to balance the load between phases the same current is always used on each of three phases.

**The long story short, even in case when 3 phases are available, the difference in power when charging current is increase by 1 amp is  (3*(230*1))=690W**
When only one phase is available the difference is 230*1=230W

Workaround:
After I have changed the value of Charger Voltage to 690V (230*3) the automation works fine:
[image]

This was the output from the FVE in the same time:
[image]

Proposal:
1. Maybe it will be better if there is another input field: 

**Number of lines**

Desc:
Number of lines provided by the EVSE aka Wallbox for the onboard charger?.
Eg: In the EU the Tesla Wall Connector and UMC gen 1 can utilize 3 phases, UMC gen2 utilizes single phase, only.

Then, the value that represent the minimal step should be computed as Charger Voltage * Number of lines

2. Change the description of Charger Voltage, to smth like:
The Line to N voltage.

flashg1, 2024-05-27:
Hi PVi1,

Thank you for the information.  Glad you found the blueprint useful.

I've updated the description in the blueprint, ie.

      description: Specify charger voltage in Volt. In Australia, single phase is 230V, and three phase is 400V due to 120 degree phase angle between phases. This can differ between countries.

Interesting that your 3 phases are in sync.  Which country is this in EU?

PVi1, 2024-05-27:
Hi @flashg1,

I am from Slovakia.

[Here](https://www.power-sonic.com/blog/single-phase-and-three-phase-voltage-by-country/) you can find more info about voltages per country.

I am sure we have 120 degree phase angle between phases in Slovakia. 

Tesla always measures voltage between Line and Neutral. That's the reason you cannot see 400V on tesla MCU. 400V is the voltage between Lines in 3phase wye(star) networks.

Today I started A/C while the car was still plugged to charger. I could see two phases were used at first and after few minutes the current on the second phase dropped to 0.
[image]
This means that car can selectively use one or more lines. But maybe that is not the case during the ac charging.

Check pages 7-9 in wall connector [manual](https://www.tesla.com/sites/default/files/support/charging/Gen_3_Wall_Connector_Manual_UK.pdf).

Then check page 10.There are power outputs based on the line type and current.

[Here](https://www.omnicalculator.com/physics/three-phase#what-is-active-or-real-power) can be found more about how apparent and active power are computed in delta and wye networks.

Based on the facts above it looks like in 3 phases networks you cannot compute one_amp_watt_step via "car_charger_voltage * 1"

In case of 3 phase networks the formula is different and is more like this:
S = √3 × Vline × Iline

Regards,
Peter

flashg1, 2024-05-31:
Hi Peter,

Thank you for the detail information.  I'll need time to digest this.

BM183, 2024-12-18:
I had the same issue with 3 phase charging causing oscillation when set to 400V, but I'm actually in Australia. Using a wall connector 3. As above, when set to 690V it now seems to work, hovers around 0-300W net grid power default settings. Thanks for the automation.

flashg1, 2024-12-20:
Thanks @BM183.  Glad it is working for you.  Added special note to this thread in the README file.

andrut1291, 2025-04-14:
Hey,

Just bought 3-phase charger. Gonna have same issue - is there any chance that these two:

[image]

are going to be based on the entities, so that I can easily change them depending on the charger I select ? 

Also since charger voltage is in my location (Poland) fluctuating between 220 V to 250 V due to poor grid quality users would be able to use 'sensor.charger_voltage' entity which is exposed for example by Tesla Fleet. This would make the calculations more accurate.

flashg1, 2025-04-14:
Hi @andrut1291 

Let's continue discussion in the [evSolarCharger project](https://github.com/flashg1/evSolarCharger/discussions/11) because I have ended support for TeslaSolarCharger.