flashg1/TeslaSolarCharger #19: Qeustion about script

gjaegy, 2024-05-29:
Hi, thanks a lot for the script, I've downloaded it this morning and it seems to work pretty well !

I had a quick look at the code, and I have a question regarding the way the charging amps value is modified. Why is that done in a loop (i.e. gradually), in 1A increment, rather than in one shot ? Is there any benefit in doing this ? 

I wonder because this means more commands sent through the Tesla API, doesn't it ?

Possibly I've misread the script, I'm not familiar with YAML yet, so forgive me if the question is stupid ;)

Thanks again for sharing your effort.

flashg1, 2024-05-31:
Well spotted.  You are right about 1A increment and decrement.

I am not an electrical engineer, but I believe a gentle ramp up or down the current is better for the power circuits than a sudden change of current.

Yes, it will take more api calls for the 1A increment/decrement.

May be someone with expertise in this area can comment on whether this is necessary or not.

flashg1, 2024-11-17:
Closing case unless this is still an issue.

closed 2024-11-17 by flashg1