flashg1/TeslaSolarCharger #22: TeslaException - Suddenly stopped working this afternoon

kimme1024, 2024-06-14:
Except for some minor hiccups which can be fixed with other automations, this blueprint has been working perfect for about 8 months now! Thanks for that!

However today the script stopped working, when I restarted it quits after some time with an error (see image below)
[image]

I fear it's on the Tesla-side of things but any idea's on how to fix this are always welcome!

Thanks!

kimme1024, 2024-06-14:
Unplugging and MCU-reset fixed the issue for now...

kimme1024, 2024-06-15:
Failed again today…
Anyone else with this issue?

flashg1, 2024-06-15:
Which version of custom Tesla integration are you using?  Are you using the Tesla Fleet API Proxy?  More info here:  https://github.com/alandtse/tesla

kimme1024, 2024-06-15:
> Which version of custom Tesla integration are you using? Are you using the Tesla Fleet API Proxy? More info here: https://github.com/alandtse/tesla
> 
> Any more clue about the exception message? Do you know which API call triggered the exception?

I’m using the latest integration
I don’t use the fleet api
Any idea on how to retrieve the extra info?

kimme1024, 2024-06-15:
Just tried resetting my MCU but that didn't solve the issue.

Checked the logs:

```
Logger: homeassistant.components.automation.tesla_solar_charger
Bron: components/automation/__init__.py:723
integratie: Automatisering (documentatie, problemen)
Eerst voorgekomen: 09:46:46 (6 gebeurtenissen)
Laatst gelogd: 16:05:05

While executing automation automation.tesla_solar_charger
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/automation/__init__.py", line 723, in async_trigger
    return await self.action_script.async_run(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1731, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 451, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 503, in _async_step
    self._handle_exception(
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 533, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 501, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/trace.py", line 284, in async_wrapper
    await func(*args)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 952, in _async_repeat_step
    await async_run_sequence(iteration)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 856, in async_run_sequence
    await self._async_run_script(script)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1210, in _async_run_script
    result = await self._async_run_long_action(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 699, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1731, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 451, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 503, in _async_step
    self._handle_exception(
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 533, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 501, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1016, in _async_choose_step
    await self._async_run_script(script)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1210, in _async_run_script
    result = await self._async_run_long_action(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 699, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1731, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 451, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 503, in _async_step
    self._handle_exception(
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 533, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 501, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1016, in _async_choose_step
[... 252 lines cut ...]
  File "/usr/local/lib/python3.12/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.12/site-packages/tenacity/_asyncio.py", line 61, in __call__
    result = await fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/teslajsonpy/controller.py", line 1390, in __post_with_retries
    return await self.__connection.post(command, method=method, data=data, url=url)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/teslajsonpy/connection.py", line 166, in post
    return await self.__open(url, method=method, headers=self.head, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/teslajsonpy/connection.py", line 218, in __open
    raise TeslaException(resp.status_code)
teslajsonpy.exceptions.TeslaException

```

```
Logger: asyncio
Bron: components/websocket_api/http.py:155
Eerst voorgekomen: 12:49:18 (8 gebeurtenissen)
Laatst gelogd: 13:57:38

socket.send() raised exception.
```

flashg1, 2024-06-15:
Is there a config to stay with the Owner API?  Or is that the default?

To stay with the Owner API, you might need to reauthenticate, or try revert to your previous version of Tesla integration.

kimme1024, 2024-06-15:
> Is there a config to stay with the Owner API? Or is that the default?
> 
> To stay with the Owner API, you might need to reauthenticate, or try revert to your previous version of Tesla integration.

I never had to reconfigure when the fleet api was released.

I’ll try to reauth and report back.

kimme1024, 2024-06-15:
Removed and readded integration.
Only thing I had to do was not to select fleet api so I guess this is the standard way.
Sadly reauth didn’t fix the issue.
Car seems to be read only…

kimme1024, 2024-06-16:
Weird, this morning. Without changing anything it started working again…

stieler-it, 2024-07-12:
I have setup the automation today, the car charges with full power, beyond what would be available as excess solar. In the trace I also see a TeslaException. Will keep an eye on it.

kimme1024, 2024-07-12:
> I have setup the automation today, I also see a TeslaException with the following stacktrace:
> 
> ```
> ...
>   File "/config/custom_components/tesla_custom/number.py", line 91, in async_set_native_value
>     await self._car.set_charging_amps(value)
> ...
>   File "/usr/local/lib/python3.12/site-packages/teslajsonpy/connection.py", line 166, in post
>     return await self.__open(url, method=method, headers=self.head, data=data)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.12/site-packages/teslajsonpy/connection.py", line 218, in __open
>     raise TeslaException(resp.status_code)
> ```
> 
> This is for Tesla custom addon, without proxy, MY 2024.
> 
> When I decrease the charging power manually it works but something (the script?) sets it back to full power.

Maybe something wrong with the sensor that measures the excess solar? W instead of kW? Had this issue in the beginning as well.

stieler-it, 2024-07-12:
> W instead of kW?

Thank you, converting W to kW in my grid sensor helped fixing the TeslaException. Automation now keeps running, triggers force update from time to time, so far it did not change charging power, even if 2kW are pulled from the grid. But that's a good point to start debugging from, thank you.

stieler-it, 2024-08-02:
Ok I re-tested the automation today and my observations boil down to a improvement idea. I think there is kinf of a rate limit around Tesla API and calling it too often leads to the TeslaException and the script being terminated (leaving everything in an undefined state).

What I observed today:

1. Car was charging with 13 Amps (max power on outlet)
2. Sun power dropped rather quickly
3. 13:09:32 charger amps decreased to 12
4. 13:09:37 charger amps decreased to 11
5. 13:09:43 charger amps decreased to 10
6. 13:09:48 charger amps decreased to 9
7. 13:09:53 charger amps decreased to 8 <- this fails with a TeslaException

So the automation tried to send at least 5 calls to the API in 21 seconds.

Now I have two ideas how this could be improved:

1. Change the logic to go from 12 to say 4 amps immediately. All data should be available to calculate the best charging amps at any point without going there step-by-step
2. Configure a local rate limit or simply loop every 30 seconds instead of every 5. I will try to change this in the blueprint and check if it fixes my setup

flashg1, 2024-11-14:
5 calls in 21 seconds is a lifetime.  I doubt anything can work with 50 commands per day.  I have been using this for over a year now with no issues.  WIFI network stability and cloud connectivity can cause Tesla API exceptions.

perjmolsen, 2024-11-14:
Hi

Just a tip from me :)

You can comminucate diretly with your tesla thru bluetooth.

I have flashed a wemos d1 mini board with tesla ble - then I am able to adjust charging amps directly from HA, and keep all my traffic local.
T
[image]

Tesla ble can be found here: https://github.com/yoziru/esphome-tesla-ble

perjmolsen, 2024-11-14:
You are welcome :) 
Tesla BLE works perfect  with Tesla Solar charger script.

flashg1, 2024-11-14:
> Hi
> 
> Just a tip from me :)
> 
> You can communicate directly with your tesla thru Bluetooth, I am using it on my Tesla model 3 highland.
> 
> I have flashed a wemos d1 mini board with tesla ble - then I am able to adjust charging amps directly from HA, and keep all my traffic local.
> 
> ![Tesla BLE](https://private-user-images.githubusercontent.com/126662740/386157501-efae8210-afb4-4f64-ba05-f0c617e96186.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE2MjczNTEsIm5iZiI6MTczMTYyNzA1MSwicGF0aCI6Ii8xMjY2NjI3NDAvMzg2MTU3NTAxLWVmYWU4MjEwLWFmYjQtNGY2NC1iYTA1LWYwYzYxN2U5NjE4Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExNFQyMzMwNTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMmRlOTc5MmRkOWNmNWJiYWY0NTU3MzlhOTc1NWI0MWQyYmNhNzIzZWYyOTU4Y2JkYjQxNWMwZDU5NWMyMWEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.aw53kvZ3grYQziIVZjR6emKPipHjCzaaA0yzHtiJSXo)
> 
> Tesla ble can be found here: https://github.com/yoziru/esphome-tesla-ble

Thank you for the tip.

flashg1, 2024-11-17:
Closing case unless this is still an issue.

closed 2024-11-17 by flashg1