flashg1/TeslaSolarCharger #23: Error: TypeError: '>' not supported ...... issue, what is wrong?

rgroothuis, 2024-08-21:
I'm installing the Blueprint and configuring it for the first time. Still learning and checking if everything is set correctly. But I got the following error message: 

Error: TypeError: '>' not supported between instances of 'str' and 'int'

See also included screenshot. And the script is not running anymore (I guess) because the Amps are not changing. 

[image]

And this is my Blueprint configuration so it can be checked if the configuration is done correctly. Segolene is the name of my Tesla Y. 

[image]
[image]
[image]

flashg1, 2024-11-17:
This response is bit late, but late is better than never!

Suggest to click on the "Changed variables" tab and look at the values under iteration 3.  There might be some clue as to which variable is causing this error.

flashg1, 2024-11-18:
Suspect issue might be related to how you set up your inverter "Grid Power Net" sensor. Please check out this [FAQ](https://github.com/flashg1/TeslaSolarCharger/wiki/FAQ).

Closing case unless you have further comments.

closed 2024-11-18 by flashg1