flashg1/TeslaSolarCharger #25: Allow for power offset to be populated from an entity

ddeconin-gh, 2025-01-02:
Hi,
It would be great if the Power offset could be populated from an entity. 
D.

flashg1, 2025-01-07:
Thank you for your suggestion. Could you please give me some background on how you plan to use this entity?  How you plan to feed the entity?  Would you be changing the value manually or from a sensor?

ddeconin-gh, 2025-01-07:
An automation would modify these values ( probably 2-3 times a month) depending on the 'capacity peak' I've had so far in the month. Capacity Peak is a Belgian (Flemish) measure of how much electricity you consume at the same time during the month. At the beginning of the month , that value would be low (since I haven't had much concurrent power use yet) but once the peak spiked to x I have more 'room' to be used by the charging for the remainder of the month.
So as long as the blueprint would use read entity that I populated when it starts running that would be fine.

flashg1, 2025-01-09:
Thank you for the explanation. Let me think about this and get back to you.

flashg1, 2025-01-31:
Hi @ddeconin-gh 

Please try latest release with your required feature.  Need to define new helper number or template sensor and set in Blueprint accordingly.

https://github.com/flashg1/TeslaSolarCharger/releases/tag/v2.1d

ddeconin-gh, 2025-02-12:
HI, that one works perfectly for me, thx!

closed 2025-02-12 by ddeconin-gh