flashg1/TeslaSolarCharger #26: Fast 'step down' A and slow 'ramp up'

ddeconin-gh, 2025-02-12:
Hi,
I'm using your blueprint with nice success. One addition that would be nice would be a different interval for ramping down (so high to low A) from ramping up.
I'm using the default 60s interval, but I see a lot of ping-ponging in the charge rate. 
Ofcourse I want it to quickly ramp down when other household appliances use a lot of power suddenly, but the ramp-up should probably be more slow to avoid the ping-ponging..
(with the Tesla Fleet API charging now for API calls, reducing the number of calls to change the charging speed would be useful.
Thx!
D.

ddeconin-gh, 2025-02-12:
[image]

flashg1, 2025-02-13:
By default, the script checks the power consumption approx. every 60 seconds and adjust the current accordingly.  There is no option to delay the increase of current, except may be to increase the cycle from 60 seconds to may be 90 seconds or higher.

If you are concerned about current increase and want to be more conservative with the increase, one thing that you can do is to adjust the skew value in the script.  It does not stop the ping-ponging of current (since that is dependent on the weather and your use of appliances), but it shifts the import/export curve to reduce the chance of importing from grid.  Unfortunately, it also increases the chance of exporting to grid.

Both skew adjustment and current stepping option are available as user inputs in my new script [evSolarCharger beta release v3.0beta1](https://github.com/flashg1/evSolarCharger).  You might want to give this a try to see if it meets your requirement.  I've added a few more user inputs and "Use Tesla API" option to work the same as TeslaSolarCharger.

I have stopped making changes to TeslaSolarCharger and will soon let people know.  New changes will only be added to evSolarCharger.